pkucmus/EVA #7: Anycubic / Creality Cartesian Mods

DavaD85, 2020-05-02:
Hi
I would like to know if is possible to have a simple mod to fit this Extruder to a cartesian printer like Chiron or Cr10, I need to have a single belt.

Thanks!

pkucmus, 2020-05-02:
We talked and there is a difficulty coming out from the belt issue CR-10 and it's derivatives has. Additionally there is not much space on the back for an EVA back belt mount. 

Ideas are welcome :)

pkucmus, 2020-05-07:
I have a cartesian version in the workings but it will require a change in the belt paths on the CR-10 deriatives

pkucmus, 2020-05-10:
The cartesian version of EVA is uploaded, more information here: https://pkucmus.github.io/EVA/sub_assemblies/cartesian_back/

pkucmus, 2020-06-06:
Cartesian back is available. Different mods for different printers would need to be done as mods for those printers. There's also V-cast incoming: https://www.ratrig.com/3d-printing-cnc/3d-printer-kits/complete-kits/rat-rig-v-cast.html

closed 2020-06-06 by pkucmus