pkucmus/EVA #11: Belt tension for sapphire plus

DavaD85, 2020-05-07:
Hi
My Sapphire plus have the belts in the opposite side and I need to mirror this part, it's possible? 
[image]
[image]

DavaD85, 2020-05-07:
[image]

pkucmus, 2020-05-07:
Since we started talking about it outside of GitHub it's already done :D 

https://github.com/pkucmus/EVA/blob/flipped_slider/stl/Dynamic%20-%20Slider.stl

Please let me know if it works.

DavaD85, 2020-05-08:
Hi
As you can see I need to move the belt a little 
[image]
And the part is a little lower than the original

pkucmus, 2020-05-27:
I think we talked about it. Two things can be done here:
1. Thicken the back part to offset the belt tensioner
2. Change the XY joiner part of the printer

I'd rather opt for the later one, since there's already little space in Y on that printer.

pkucmus, 2020-06-06:
Different belt tension sliders are now available - fill support for Sapphire plus is not something I can take on. Closing this one :)

closed 2020-06-06 by pkucmus