pkucmus/EVA #8: Mount for BMG Aero

KipK, 2020-05-07:
Hi

Great work you did here.

I've started to make a mount for the new BMG Aero from mellow. It' a kind of a remix of a BMG with Hemera heatsink, the titan aero way.

https://www.aliexpress.com/i/4000084805415.html

I'd like to remix your fan duct on it, would you mind to share the step file ? 

Thanks

Gui

pkucmus, 2020-05-07:
Thank you and Awesome! You'd like it to be a part of EVA? 

The thing is that I can't guarantee that stuff will not change at this point but if you still want to give it a go I could give you the required step files (the ducts and a base face maybe?).

KipK, 2020-05-07:
Sure, whatever you have that can inspire me too. If you have the Hemera
step file too as booth have almost the same footprint

I'm almost done making a BLV mount for it that fits bltouch and 5015 blower
but I'm now stuck fitting a proper cooling duct.
I don't mind redoing it by to be more EVA lookish ;)
How the fan ducts doing is job on your setup?
I was quiet used to double 4010 blowers.

KipK, 2020-05-07:
Titan aero mount seems the nearest start  for this one. So I'd rather have this step
Here is a STL of the extruder
https://www.thingiverse.com/thing:4226245

pkucmus, 2020-05-08:
To answer the question about he fan duct - it's doing very well, I was surprised i.e. how long of a bridge I can make with it.
Here's a CFG simulation result from this week:
[image]

pkucmus, 2020-05-27:
Ho is it going? Do you need anything else?

KipK, 2020-05-30:
> How is it going? Do you need anything else?

Hey.

Sorry was busy those days.
I've misthought it was for BLV mgn printer.
Tryed to modify for it and fits all my needs but finally was not satisfied
by this extruder positioning and weight. ( Not your design but this
extruder design sideways as I prefer a dual blower config.
For now I've made something else, but will probably work on it back later

pkucmus, 2020-09-05:
Thanks, I'll close this one for now, we can always revisit the idea in the future

closed 2020-09-05 by pkucmus