pkucmus/EVA #9: fan problem

DavaD85, 2020-05-07:
I need to move a little down the back fan because doesn't fit in my Sapphire plus frame.

DavaD85, 2020-05-07:
[image]

pkucmus, 2020-05-09:
I made 4 more versions which total up to 5: 
* 30 deg
* 45 deg
* 60 deg (default)
* 80 deg
* 90 deg

And I will upload those tomorrow.

[image]
[image]
[image]

DavaD85, 2020-05-10:
Did you have time to upload these stl?

pkucmus, 2020-05-10:
5 different angles are now available: https://pkucmus.github.io/EVA/sub_assemblies/core_xy_back/ also for cartesians https://pkucmus.github.io/EVA/sub_assemblies/cartesian_back/.

Please yet me know here if you are happy with them.

pkucmus, 2020-05-27:
Closing the request as resolved. Let me know if you need anything else.

closed 2020-05-27 by pkucmus