pkucmus/EVA #16: 8mm ID probes with horn duct conflict

M1dn1ghtN1nj4, 2020-05-26:
For the setup I'm looking to put together, there seems to be a problem with the both the 8mm and 12mm inductive bed probes with the horn duct.  Or could it be that I'm somehow not assembling this right.  The photos show it like it fits, but I can't find a combination that works.

[image]

pkucmus, 2020-05-27:
You are probably looking for the "_right" variants of the probes.
[image]
[image]
All probe holder variants are here: https://github.com/pkucmus/EVA/tree/master/stl/Probe%20Mounts

Please let me know if that helps.

M1dn1ghtN1nj4, 2020-05-29:
Yep, that works.  I picked the left side one because it brings the probe more in line with the nozzle.  It's how I've designed all of my printers.  The probe and nozzle are perfectly in line in the Y axis.  Just a bit OCD on my part.  I also had to shorten up the horn duct a little.  The output was about 4mm to 5mm ahead of the nozzle tip.
[image]

pkucmus, 2020-05-29:
Alright, thanks for the update! Issue resolved then?

M1dn1ghtN1nj4, 2020-05-30:
Yep.  All good!  Thanks

closed 2020-05-30 by M1dn1ghtN1nj4