pkucmus/EVA #17: Requested Mosquito/Nf-Crazy Mount

morpheus200, 2020-06-05:
Request for Mosquito or NF-Crazy (Clone) Hotend. The Hotend is almost compatible with the V6, the differences are very small.

pkucmus, 2020-06-05:
There are people trying to organize those for me so that I can measure, design and test - so it's coming, but has no ETA yet.

joaobarros, 2020-06-18:
Plus one on this request, let me know if you need anything

pkucmus, 2020-06-18:
I should get a Mosquito soon (I hope) I'll start the work then.

pkucmus, 2020-10-13:
Those are ready to be tested, once I'll do my testing I'll release, you can try to use the current version but stuff may change. Also If you'll use please share your feedback: https://cad.onshape.com/documents/1cea28d26150bba34a3d6614/w/fd185c58d38d06bda7ec83a6/e/180aabbd13bbeefb972e31a1
[image]

EduardoOliveira, 2020-10-13:
The fan face looks awesome

maziggy, 2020-10-14:
> Those are ready to be tested, once I'll do my testing I'll release, you can try to use the current version but stuff may change.

Where can I find it?

pkucmus, 2020-10-18:
Please check out the unified BMG/BMG-M approach that also gets rid of the groove mount adapter requirement. Thanks miklschmidt!
https://cad.onshape.com/documents/1cea28d26150bba34a3d6614/w/fd185c58d38d06bda7ec83a6/e/db5e779d6e295e18e56f51d3

miklschmidt, 2020-10-18:
Muuuch better! 👍

maziggy, 2020-10-18:
When using a BMG-M I don't see any advantage compared to the design before?

miklschmidt, 2020-10-18:
@maziggy There's no real benefit to the BMG-M version unless the mosquito is mounted directly to the extruder (shorter filament travel path). As far as i can tell, that's not possible in the current EVA design.

pkucmus, 2020-10-18:
EVA is much about supporting what people may have on hand, if one is switching to EVA from a Mosquito with a BMG that BMG is likely a BMG-M, so we wanted EVA to support either one of those. This particular update makes me happy since in theory one set of those parts can support both permutations. I hope to have it tested soon.

maziggy, 2020-10-19:
Thanks, makes sense. I already printed the former version and will try it today.

maziggy, 2020-10-20:
Found a little issue. Printed all parts for a mgn12 carriage and found out that the front plate doesn't fit. The screw holes are to close to each other. It looks like the top_mgn15_bmg.stl fits to a mgn12h. Printing it while writing and let you know.

maziggy, 2020-10-20:
Just fount out that your model is for a mgn12c. I have a mgn12h. Will move the screw holes.

maziggy, 2020-10-25:
Will test it now. Assembling was no issue. Very nice design.
[image]

pkucmus, 2020-10-25:
1.4.0-RC2 carries newest improvements (still work in progress): https://cad.onshape.com/documents/1cea28d26150bba34a3d6614/v/93a8541e17dd938c15833def/e/db5e779d6e295e18e56f51d3

maziggy, 2020-10-26:
> Those are ready to be tested, once I'll do my testing I'll release, you can try to use the current version but stuff may change. Also If you'll use please share your feedback: https://cad.onshape.com/documents/1cea28d26150bba34a3d6614/w/fd185c58d38d06bda7ec83a6/e/180aabbd13bbeefb972e31a1
> [image]

Are you aware of any filament path issues in this version? First print was successfull but after unloading the filament I was not able to load a new one. Filament was stuck at the end of the printed part diretly above the Mosquito. Cleared the path and again the first print was good but problems reoccured when changing filament.

miklschmidt, 2020-10-26:
@maziggy you should use a piece of bowden tube between the BMG and the Mosquito.

pkucmus, 2020-10-26:
Added some clearance to the PTFE hole, 1.4.0-RC3: https://cad.onshape.com/documents/1cea28d26150bba34a3d6614/v/630104b77ee640cfa5db0a31/e/1cca3cb2ad6c3a96d3e9486e

maziggy, 2020-10-28:
Thanks, works perfect.

pkucmus, 2020-11-01:
Mosquito support is now released in 1.4.0. Thanks for all the input!

closed 2020-11-01 by pkucmus

maziggy, 2020-11-08:
Within the last few months I've tried mayn different designs for the combination Mosquito and Bondtech. Some good ones but nothing without any disadvantages. After printing now for two weeks with your design I can say that's the way to go for me. No issues at all.

To tell a long story short: no knick-knack, very easy to assemblle and it just works reliable as it should. Many thanks for your work!!!