pkucmus/EVA #18: Mount request for Duet Mini IR led sensor 1.2

piksliviksi, 2020-06-18:
Hi!

Any chance of getting the mount for https://miscsolutions.wordpress.com/mini-height-sensor-board/ ?

pkucmus, 2020-06-18:
I think it should be possible. I don't know anything about it though. Saw some examples like: https://www.thingiverse.com/thing:939909. Is that something to aim for?

piksliviksi, 2020-06-18:
Yes, that's the thing. 
It has to be 2.5-3mm higher than the nozzle from the buildplate. Other than that, it's a fairly simple pcb with 2 holes in it: 
Top right corner (24.0, 17.62)
Mounting hole centres (2.70, 14.92) and (21.11, 14.92)
Mounting hole diameter 2.8
and the pcb thickness is 1.5mm and the total amount of space it takes is about 15mm, together with the components on the pcb.

joaobarros, 2020-12-10:
STEP model of the sensor from here: https://grabcad.com/library/dc42-mini-ir-height-sensor-1
[dc42-mini-ir-height-sensor-1.snapshot.2.zip](https://github.com/pkucmus/EVA/files/5674301/dc42-mini-ir-height-sensor-1.snapshot.2.zip)

The sensor bottom should be ~1.5mm higher relative to the nozzle as per documentation:

_"The bottom edge of the board should be no more than 2mm higher than the tip of the nozzle, to ensure that it will trigger before the nozzle touches the bed. To avoid the risk of the board fouling on the print, we suggest that the bottom edge is at least 1mm higher than the tip of the nozzle. So aim for 1.5mm."

Although a fixed position holder would be good as a start, an adjustable height holder would be great to be able to adjust to different bed surfaces.

pkucmus, 2020-12-31:
Moving to EVA 2

closed 2020-12-31 by pkucmus