pkucmus/EVA #23: XY joiner for HEVO

joaobarros, 2020-07-06:
There's a big potential crowd for the EVA system on current HEVO builds that want to move to rails.

A mod of your Easymod would allow for a "simple" migration to rails and the EVA

Some considerations about the differences of an HEVO and the Easymod on a V-Core Pro:
- 3030 vs 2020
- on the HEVO the belts run closer to the frames then on the V-Core Pro
- 
- the Easymod uses the plates from the original V-Core Pro, the new XY joiner should be all printed to have dependencies on custom hardware

As we discussed this is a personal need/request but one that I have confidence would have a large amount of followers :)

pkucmus, 2020-07-15:
Are there any STEP files for an MGN HEVO I could use as reference?

joaobarros, 2020-07-15:
This project has a .f3d file: https://www.thingiverse.com/thing:4460475
If you are unable to open .f3d files I can export.

That project has 2 versions: full (XY) and half (just X), but I believe like myself, most people will go for the full version
Thank you for looking into this!

pkucmus, 2020-07-15:
Please export to STEP, thank you.

joaobarros, 2020-07-15:
Online model: https://bit.ly/393Kht4
STEP file with all the parts: https://www.dropbox.com/s/nnhjuvkw4zrp5u2/HEVO_3030_MGN_Rail_Conversion.step.zip?dl=0
Let me know if you prefer single objects STEPs

pkucmus, 2020-07-15:
This is great, thank you!

pkucmus, 2020-12-31:
Please move this Issue to EVA 2 https://github.com/EVA-3D/eva-main/issues if it's still something to desire :) (please also put the link to this issue there)

closed 2020-12-31 by pkucmus