pkucmus/EVA #24: Duet 3 Toolboard Support

philkry, 2020-07-16:
Amazing design you came up with, like it a lot!

For my new CoreXY printer i'm planning to make use of the new Duet 3 Toolboards (https://duet3d.dozuki.com/Wiki/Duet_3_Tool_Board) to reduce the number of cables running to my extruders.

Have you considered supporting it on your EVA platform?

It can be mounted onto a Hemera, so it can be used directly with only minor changes to Z probe mounts.
However for the other extruders, i'm unsure how to integrate it.

philkry, 2020-07-21:
Hey, i spent some time tweaking the V6 BMG version. It is not ideal yet and requires more tweaking as it sticks out to the side a bit too much.
[image]
[image]

pkucmus, 2020-07-21:
Nice. The drawback is that the motor size is limited now but I'm happy that you put work in to EVA. Thanks!

To say that EVA can support the tool board I would need to add that capability to all variants - not sure it's feasible.

philkry, 2020-07-21:
Yes, i was considering the stepper size as well, but thought when using a geared BMG extruder, you most likely will only require a pancake stepper rather than a full sized one. It would also make the overall assembly considerably more heavier.

I agree, its probably not really feasible to integrate it into all mounts, maybe using a Toolboard is even too niche but i thought i share this as i plan to build my new printer from scratch based on them (but i also only plan to use Hemera and V6 BMG).

I will post updates here in case someone else finds it useful.

pkucmus, 2020-07-21:
Awesome. Let me know when you are done and would like to be featured in EVA's documentation.

philkry, 2020-08-10:
so i did some changes to my initial design. i did move the mount to the side of the top module. this now allows all stepper sizes to be used, but more importantly, allows quicker changing the entire carriage. 
with the toolboard mounted at the back (first iteration) you needed to disassemble the entire thing to get the motor out, which kinda defies the purpose of using a toolboard entirely.

you can find my version (mgn15 only for now) here https://github.com/philkry/EVA/blob/master/stl/Tops/top_MGN15_bmg_toolboard.stl

please note, it is still untested as i'm waiting for my printer to be delivered ;)

[image]

philkry, 2020-08-24:
@arsi-apli nice design! 
just wondering, is there a reason why you did not just use the Hemera Alternative version of EVA? It would allow to mount the toolboard directly onto the front of Hemera, just the probe mount would need adjustment. that way you could stay as close to 'stock' EVA as possible.
dont get me wrong, it looks good, although more complicated than i think it needs to be.

pkucmus, 2020-08-24:
There are a few things to consider:
1. What is the benefit of having the tool head? Having it on a fixed Hemera carriage is just reducing the cables required from the main board to the carriage, right? Quite frankly it does not seem compelling. (more in point 3)
2. To be a part of the EVA platform a feature needs to work for all variants, the top part feels like a more proper or generic place to mount this.
3. Let's look into the crystal ball a little bit here: if EVA was to support tool changers, it would be a face with the locking mechanism that would grab a face that would hold a tool board, maybe? I don't know yet.

philkry, 2020-08-25:
> What is the benefit of having the tool head? Having it on a fixed Hemera carriage is just reducing the cables required from the main board to the carriage, right? Quite frankly it does not seem compelling. (more in point 3)

well, yes. the reduced amount of cables was my initial reason to opt for a toolboard. i like the EVA platform as it would allow me to reduce the efforts needed on manual changing the tool heads. less cables help here as the EVA carriage would be pretty much self-contained and you could just connect power + CAN bus.

on the other hand, you could achieve the same by crimping connectors to all cables (2 fans, thermistor, heater, probe) but that makes it bulky and not so easy to use again. if EVA ever makes it towards a tool changer system (your point 3) that point would even become obsolete as you would have the tools permanently connected. 

were i still see benefits though, even on the long term is, that the tool board also provides you with an additional stepper driver for your extruder. you would not need extension boards anymore and could again achieve self-contained EVA carriages which contain everything you need for driving an extruder assembly. 

> To be a part of the EVA platform a feature needs to work for all variants, the top part feels like a more proper or generic place to mount this.

good point! makes sense.

joaobarros, 2020-08-25:
I too am interested in this, the toolboards will become more mainstream as they will be supported in the upcoming Duet 3 Mini
High-end heaters and temp sensors come with short cables so board placement and thus distance is important, just wanted to point that what 😉

arsi-apli, 2020-08-25:
My opinion is that the toolboard helps to reduce vibrations. A common bundle of cables already weighs something, plus the load they exert on the printhead is not linear. That's the main reason I want to use it. And other benefits that have already been mentioned. 

Unfortunately, there are belts by the sides and therefore I'm still thinking about putting the linear rail down and a flat cable for power supply and CAN Bus will be stored in the upper slot in the ALU profile with a short loop to the print head. The same mounting will be on the second axis.

pkucmus, 2020-08-25:
Thank you all for the valuable info. I like the idea of supporting the tool board (if that was not obvious). 
I think we need to look at two permutations: 
1. for the top free heads (bowden, aero, hemera) - the tool board flat on top which will not work best with the cable mounts coming in 1.3.0
2. for the top motor ones (titan, bmg) - the tool board behind the motor?

Overall this makes me think about extra holes everywhere - as mounting points for anything extra that people may think about, i.e. "EVA Standard Expansion" screw holes. Then a tool board would just need a mounting part that is compatible with the board on "one side" and with the EVA Expansion holes on the other. It may just be the main feature of EVA 1.4.0.

P.S
EVA 1.3.0 is coming this week

pkucmus, 2020-08-28:
EVA 1.3.0 is now released please check if it changes anything for you. Maybe you could leverage the new back mounting holes somehow (just a hot guess).

pkucmus, 2020-08-29:
Played around with it a bit and got this as a result:

[image]

The top heavy ones (BMG and Titan) would have the vertical variant and the top free ones (bowden, Aero, Hemera) would get the horizontal variant.

TODO:

- Horizontal and vertical need a mounting point for a cable holder
- The tops will need to change to take the screw that will firmly hold the bottom part in place
- Vertical ones need another anchor point (not to rock like crazy when moving in Y)

What do you think? 
https://cad.onshape.com/documents/1cea28d26150bba34a3d6614/v/e72f5d6038b183a8b63f5cd0/e/44f6d805bfa7d15742b3ac6f
(check the assembly aero and assembly BMG files, toggle the configuration in top left to change the variants)

philkry, 2020-08-30:
Wow, amazing. The cover looks really sleek and is perfect spot for your new logo ;) 

I will test this and give feedback as soon as i built up my printer!

pkucmus, 2020-08-30:
@philkry please consider this a beta test version. At least what is in the TODO needs to be addressed before we call it done :) I'll give it more time in the evening.

pkucmus, 2020-08-30:
I've added cutouts for the LEDs and other SMD components that are sticking out close to the case wall
[image]
But I can't wrap my head around the cable mount or the vertical variant, I'm afraid it's going to wobble like crazy.

pkucmus, 2020-09-01:
Can someone please confirm how flat the bottom surface of the board is? A picture would be nice. The original STEP has all the THT legs but I'm wondering if there's anything else.

philkry, 2020-09-01:
took a two quick pictures. can make more if needed. there are only THT legs, the longest ones are the one for the power connector. their length varies between 1.7-2.2mm on my boards.

[image]
[image]

pkucmus, 2020-09-05:
Ok, I think I'm at a release candidate point. 
For flat top variants:
[image]
For BMG and Titan we would leverage a bridge over the motor that will hold the ToolBoard case steady
[image]
[image]

https://cad.onshape.com/documents/1cea28d26150bba34a3d6614/v/47bcc210604fef2704e4d8dc/e/e089ef52b1a648887b30ed21

Please let me know what you think

joaobarros, 2020-09-06:
I love it 👍 
My only concern is traping heat on the back of the board, specially the stepper.
All Duet boards are designed as the bottom side of the boards as the hot side.

philkry, 2020-09-09:
i like it as well! looks rock solid.
not sure how hot the stepper actually gets? they designed the toolboard to be mounted directly onto a hemera extruder. there  the heat also gets kinda trapped + it gets additional heat from the stepper motor.

@pkucmus i somehow have trouble exporting the back part. not too familiar with onshape, but if i select the back and click export, i do get the part as STL, but the screw holes for the toolboard mount are actually filled/solid.

pkucmus, 2020-09-10:
@philkry that is desired, you will need to drill or poke those holes after print. Put that in a slicer and it'll make sense:
[image]

philkry, 2020-10-02:
@pkucmus i printed this in the meantime and can confirm it fits perfectly 👍 
[image]

pkucmus, 2020-10-02:
Awesome, although @joaobarros raised an issue that the screws that are under the tool board may short the pins of the board. No bad experiences there? or did you use some short headed screws?

philkry, 2020-10-02:
no shorts, the board is just working fine. but to be honest i did not pay too much attention to the screws under the board (normal hex screws), maybe i'm just lucky? 
i don't really wanna take it apart again now to verify, getting the cabling nicely done was too much work :D

philkry, 2020-10-04:
i just disassembled everything and double checked. and yes, i agree, the motor connector is dangerously close to the screw. i think i was just lucky with my particular toolboard. 
maybe you can make the base a few millimeters thicker so there is a bit of a safty distance?

pkucmus, 2020-10-10:
As a FYI I think I'll move this out of EVA and put it one some "EVA extras" project, I'm trying to keep up with all the permutations and it's a chore. Any ideas for a better name than "EVA extras"?

philkry, 2020-10-12:
Guess EVA extras would be fine, other options would be 'EVA extensions' or 'EVA contrib' ?

EduardoOliveira, 2020-10-12:
I can see contrib for parts that are designed by other members and of which
you only have the step file, linking to external sources, and not
garanteing updates

pkucmus, 2020-10-12:
Someone suggested "EVA Addons" on Discord, I think that represents that those would be best. Toolboard addon is now in a separate document: https://cad.onshape.com/documents/255cffe335d549be4c0d4c52/w/027a43fe00d211ea4230f02e/e/821b45343fc77d009d4ac8c6 (still work in progress) but once I release everything will be described in main EVA docs

kuon, 2020-10-20:
But how will you "standardize" the addons mouting in the base EVA? 

Should we consider those 5 holes "addon mount"? But what about variant without the top part (like BGM bowden)?

[image]

pkucmus, 2020-10-20:
The 3 hole mount would be a standard. 
For the variants that have their tops free there's doing to be an angled tool board base (so the toolboard will be flat on top). 
The "top heavy" ones get the 2 extra holes for the board base to grab by so that the tool board does not wobble like crazy on Y moves.

kuon, 2020-10-20:
So if we want to design other addons, we can rely on the 3 holes for variants with no top, and 5 holes for those with "heavy" top?

pkucmus, 2020-10-20:
I want to say "yes" but I will be more comfortable once I figure this one completely, the 3 holes are set in stone I think, the 2 top ones not so sure, and for the ones with clear top there's an opportunity for a screw or two as well
[image]

kuon, 2020-10-26:
The bracket for the BMG has a very thin wall (red line). I think it should be around 0.4-0.5mm for better printability.

[image]

joaobarros, 2020-10-29:
So I finally printed my Toolboard cover and I've got some issue, the major one being the extra height of a connector:

[image]

Then it's small details that are difficult to print and can eventually break off, as I broke 2 while clearing the holes:
[image]
[image]

This part here could be a bit wider:
[image]
[image]

This piece here could be wider or even not there at all:
[image]

This details here are difficult to print and borderline, any LED that's soldered a bit off will be an issue. Does it makes sense to have them? 
[image]
[image]

This piece here could be wider or even not be there at all:
[image]
[image]

I see @philkry has by the photos the same board version as mine v1.0 and the extra height connector and he says everything fits perfectly which I find very strange?!

And a design challenge:
In order to change something attached to a screwed connector we would have to unscrew 5 screws, 1 of the cable holder and 4 for the cover.
What if the cover is not so pretty, so square but has cutouts so one could screw/unscrew without having to unscrew the hole cover?

philkry, 2020-11-01:
@joaobarros it could be my slicer settings that made it fit better for me as i was working with negative horizontal expansion to compensate for too narrow holes. 

> This details here are difficult to print and borderline, any LED that's soldered a bit off will be an issue. Does it makes sense to have them?
i would say yes. at least the red led is helpful to see if CAN communication between the two boards works fine. it helped me quite a lot when assembling my printer.

>In order to change something attached to a screw connector we would have to unscrew 5 screws, 1 of the cable holder and 4 for the cover.
i agree, this is not ideal currently. it looks very nice but at the cost of practicality. 
i would also favour a version which would expose the terminals, so we could access the screws without disassembling the entire thing.

joaobarros, 2020-11-01:
@philkry I understand what you're saying about the slicer settings but that doesn't explain the plug I show on the 1st picture, that's some 2mm at least of extra height.
I'd love to see a photo of that side on your print and Toolboard

pkucmus, 2020-11-01:
@joaobarros I made changes according to your (amazing) feedback, took a different approach with the bridge mount
https://cad.onshape.com/documents/255cffe335d549be4c0d4c52/w/027a43fe00d211ea4230f02e/e/795e637b64bf65caed4dccd1?configuration=default
[image]

pkucmus, 2020-12-31:
Moving this to EVA 2 :)

closed 2020-12-31 by pkucmus