pkucmus/EVA #27: Compact Direct Drive / short filament path option

hernancurras, 2020-08-19:
Hey guys... I have been following the EVA project since long time ago. And to brainstorm EVA modular design, I would like to build/have a double geared compact option inspired on the Bondtech BMG Prusa Mk3 / MZ76v2 / Voron Afterburner 2.4 / CBRExtruder / ...

https://www.prusaprinters.org/es/prints/17584-bondtech-bmg-prusa-mk3smk25s-v2-fdm-printable-rhd-
http://vorondesign.com/voron2.4
https://github.com/AL3DAN/CBRExtruder
 
My personal taste, is build.in - self contained BMG dual geared + E3Dv6 + BLTouch.

Is it a feasible branch on EVA project?

pkucmus, 2020-08-20:
If I understand the idea I think I like it. Would give one the benefits of a BMG without the BMG (?). 
With EVA I think there is an opportunity to do it "better" as in the fan is already taken care of and the motor can be on the back (saving Y space and moving the motor mass to the back). So it would boil down to making an EVA face I think. 
That would be very nice.
All the source will be published with the next (1.3.0) release which is very close.

kuon, 2020-10-20:
Isn't the hemera a good option if you want a short filament path?

pkucmus, 2020-12-31:
Please move this Issue to EVA 2 https://github.com/EVA-3D/eva-main/issues if it's still something to desire :) (please also put the link to this issue there)

closed 2020-12-31 by pkucmus