pkucmus/EVA #32: Use in BLV MGN 12

cjeannerot25, 2020-08-29:
I have a problem for use with BLV MGN, the connection part of the belt is reversed, it should be top right, and bottom left, currently on the EVA project it's reversed. I also use MGN12H and it's MGN12C on the EVA project (I redid the part for the BMG direct drive)

pkucmus, 2020-08-29:
BLV support was never tested, what if you'd mirror the back part? Not sure if the belts will align though

cjeannerot25, 2020-08-30:
[image]

I confirm belts is aligned ;)

cjeannerot25, 2020-08-30:
For back it's possible to connected [image]

pkucmus, 2020-09-05:
Please see: https://cad.onshape.com/documents/1cea28d26150bba34a3d6614/v/0010d118035086ce95d3908d/e/8c43124770d234ffb36295f8 (notice configuration in top left) - far from being done since this only works for the V6 based variants but for BMG testing purposes should be ok

cjeannerot25, 2020-09-05:
No it's not possible to use in BLV mgn.

pkucmus, 2020-12-31:
Moved to EVA 2

closed 2020-12-31 by pkucmus