pkucmus/EVA #34: Dragon hotend support

pkucmus, 2020-09-13:
Seems like it's working out of the box :) At least dimension-wise (thank you Jonh D. for the pictures and testing!).
I'll leave the issue here in case there are more test results -stuff like: making sure nothing melts, is the air flow adequate, etc.

If there's anything to share in the topic please let me know here.

[image]
[image]
[image]
[image]
[image]

philkry, 2020-10-03:
@pkucmus i'm also using the dragon hotend but but my heater has issues keeping up the temperature.
it is stable until the part cooling fan turns on, then it goes down by 10-15° and never recovers. 
i'm wondering if my heater is bad (brand new E3D high precision 30w) or if the fan duct somehow blows onto the heater cartridge? 
did you experience something similar while testing? if not i will try replacing my heater.

pkucmus, 2020-10-03:
Horns duct is made not to blow directly on the nozzle or the heat block: 
[image]
that said there is a lot of air being pushed everywhere - will bounce of the bed or the print, etc. 30W seems pretty weak (I'm using 50W ones), I had similar issues on different printers and 40W heaters, a silicone sock does a good job of isolating the cool air as well. 
Before replacing your heater please run PID tuning and maybe PID tuning with the fan 100% PWM on - to teach it how to properly heat itself under those conditions.
I hope this helps.

philkry, 2020-10-03:
silicon sock is mounted already and the heater is also PID tuned with having part cooling at 100%.
i have a spare 40w heater here, will try this one and see if it improves. my first guess was just the horn duct, since i use the same heater model and part cooling fan on my Hemera extruder with super stable temperatures.

pkucmus, 2020-12-31:
Done in EVA 2.1.0: https://github.com/EVA-3D/eva-main/issues/5

closed 2020-12-31 by pkucmus