pkucmus/EVA #37: Idex version fan mount is too fragile

kuon, 2020-10-03:
[image]

The highlighted element is quite fragile and should be made thicker.

pkucmus, 2020-10-03:
I have thicken the "hooks" by 1mm each, for IDEX and Cartesian Backs, please see: https://cad.onshape.com/documents/1cea28d26150bba34a3d6614/w/fd185c58d38d06bda7ec83a6/e/8d4e73fd9c11258ba0412b03

kuon, 2020-10-03:
Yeah, that should do it.

kuon, 2020-10-06:
There is another issue with those two hooks, they are separated by 15.2mm, but my fans are 15.25mm. I looked at different fan datasheets, and the tolerance is between 0.25mm and 0.5mm, so I suggest we change this width to 15.5mm instead of 15.2mm.

pkucmus, 2020-10-07:
Done, please let me know what you think: https://cad.onshape.com/documents/1cea28d26150bba34a3d6614/w/fd185c58d38d06bda7ec83a6/e/8d4e73fd9c11258ba0412b03

kuon, 2020-10-07:
Yeah, looks good to me.

closed 2020-10-07 by kuon