pkucmus/EVA #39: Aero face modelling glitches

mpeto, 2020-10-26:
Extrude 49 - protrudes on the other side - esthetic issue
Extrude 41 - wrong profile selection, hexagon's inscribed circle smaller than original hole's counterbore diameter (same issue with all faces with nut sockets)
Chamfer 26 leaves a small gap at the intersection with the motor mounting face

pkucmus, 2020-11-22:
I lost the CAD link from you, can you provide me one? Or the STEP file

mpeto, 2020-11-23:
I think this was it:
https://cad.onshape.com/documents/00da96811a434ceb4ebdf5e1/w/e6de0b894be33ca234b23969/e/79c5b4f601dfed4a447bd261?configuration=List_kYc0XMLK5ouBl4%3DDefault
Onshape<https://cad.onshape.com/documents/00da96811a434ceb4ebdf5e1/w/e6de0b894be33ca234b23969/e/79c5b4f601dfed4a447bd261?configuration=List_kYc0XMLK5ouBl4%3DDefault>
Login / Signin to Onshape - the modern CAD system used by thousands of companies worldwide.
cad.onshape.com

________________________________
Feladó: Paweł Kucmus <notifications@github.com>
Elküldve: 2020. november 22., vasárnap 16:16
Címzett: pkucmus/EVA <EVA@noreply.github.com>
Másolatot kap: mpeto <matyas.peto@hotmail.com>; Author <author@noreply.github.com>
Tárgy: Re: [pkucmus/EVA] Aero face modelling glitches (#39)

I lost the CAD link from you, can you provide me one? Or the STEP file

pkucmus, 2020-12-31:
Aero was remade in EVA 2. Please feel free to comment and recommend improvements https://github.com/EVA-3D/eva-main/issues

closed 2020-12-31 by pkucmus