pkucmus/EVA #43: Top part only for MGN12C

alessg1, 2020-11-10:
Your top part is only design for MGN12c witch have 15mm x 20mm distance between holes.
You shoud also make version for MGN12H witch have 20mm x 20mm

EduardoOliveira, 2020-11-10:
Wich top part do you need? I've edited the tops and printed the BMG, I know the bowden top is also ok, not sure about the other one, I can share if you like. Mng12h

alessg1, 2020-11-10:
I already modify BGM top to be compatible with MGN12H. But now i need to modify bottom one to be compatible with BLV mgn cube endstop

pkucmus, 2020-12-31:
Please move this Issue to EVA 2 https://github.com/EVA-3D/eva-main/issues if it's still something to desire :) (please also put the link to this issue there)

closed 2020-12-31 by pkucmus