pkucmus/EVA #47: Titan v6 front question

LucidityCrash, 2020-12-24:
Why does this not place the v6 hotend inside the Titan but clamp it outside lower down ?  This makes no sense you've essentially increased the filament path length making giving it all the issues of a bowden extruder but not the advantage of less moving mass.  It also increases the height of the assembly meaning a reduced Z build volume.

pkucmus, 2020-12-25:
It's hard to answer a "why it's done that way" question... There are many reasons and factors in which had be considered:

1. EVA is about being modular - the extra space often asks as a compatibility layer between different variants and permutations
2. Moving the tip of the nozzle as high for it to go into titans groove mount would make the nozzle be only 0.4mm bellow the 2020 - not an option
3. I wanted the motor (the heaviest part) to be as close to the center of the carriage as possible - that defines where the motor is and where the titan is.
4. I also wanted to have the layer cooling duct not obstructing the view of the nozzle - which either meant having the duct come from bellow of the top, bellow was better for easier parts (to design, maintain and print) 
5. **24.7 mm added filament path** (in the case of titan and V6) has very little to do with a hundreds of mm long bowden tube, I was using EVA Titan for the longest time and those extra mm never bothered me.
6. The build volume of for e.g. Rat Rig V-core Pro printer was increased or at least did not change 

[image]

LucidityCrash, 2020-12-26:
Understood ... You could have done what you did for the Aero, which is essentially the same but even shorter and rotate the motor by 90 Degrees and place it in front of the profile.  Obviously this violates point 3 :) 

Do/Did you use the groove mount bowden adaptor or just run the tube straight up into the black plastic part inside the titan.

Out of interest what did you change to after the Titan ... I'm in the process of upgrading my Prusa i3 to the RatRig V-Core Pro and originally I was going to take the extruder/hotend off but now I'm thinking I'm going to get a new one for the new printer.  The Hemera looked good but is sold out ... was thinking of going with the Aero.

pkucmus, 2020-12-26:
You don't need the groove mount, the PTFE has a place to "lock" itself in place inside the white, slippery part of Titan. But I made a printable one in case someone would still opt to use one.

I got a BMG from the community so I'm using that on my main V-core 3, my V-core Pro will get a Dragon + Titan. But It have no need for the short path ones like Aero or Hemera (those two are also front heavy and steal from of the build space). If you have a Titan I'd try that - you may end up being perfectly happy with it. 

There's also an idea to make a short path. Prusa MK3-like EVA face with dual gear extrusion but that's in the future.

pkucmus, 2020-12-31:
I'm assuming there's not much to discuss here. Closing, thanks!

closed 2020-12-31 by pkucmus